parezj/EMBO #4: Great project - thank you

janjos0, 2022-01-26:
Hi Jakub, I have no issue but just wanted to congratulate you on this great project.

I flashed the .bin file in a "BluePill", downloaded the zip file and extracted the Windows EMBO front end, clicked on Scan, selected the COM port, clicked Connect and everything works fine. So easy.
The available instruments are very useful and the GUI looks very professional.

I do have a question, will the the F303RE bin file work on a F303RC ?
I am thinking of buying this Olimex board:
https://www.olimex.com/Products/Duino/STM32/OLIMEXINO-STM32F3/open-source-hardware

And also a remark, maybe you could use the built-in green LED to like flash every x time, instead of just ON when connected.
I am thinking something really basic like in the main loop if(!ledflash--) toggle_led(); ledflash = x;
This way it could tell the user that it is connected and active/doing something.
Only ON can also mean the STM32 pin is pulled to GND (in the BluePill).
This is surely not a must, as I said everything works fine just like it is and this is a great project.
I wish you all the best.
Janjos

parezj, 2022-01-26:
Hi Janjos0.

Thank you very much for appreciating this project, I am so glad you like it. 

I compared F303RE to F303RC and there are some small differences, but the huge one is RAM size, which is smaller in F303RC. So the bin file will definitely not work, sorry. I will open an issue to create port for F303RC and I will do it, when I get to it. Olimex board looks really nice, so it will be nice motivation.

If you dont want to wait, you can buy Nucleo F303RE or you can fork this project and create new .ioc CubeMX configuration + .cfg header configuration, but as there is not any tutorial yet, it could be pain.

Funny thing about blinking LED is, that it WAS my first intention. If you connect board to PC and send 1 SCPI command from terminal, it does really blink. But when it is connected to PC app, communication is so fast, that it looks like the LED is just ON. I will definitely work on this issue. Thank you!

janjos0, 2022-01-26:
Hi Jakub,

Thanks for considering porting your project also to the F303RC.
I thought the F303RE bin would not work because of some peripheral configuration, did not expect the problem to be RAM related, 40kB vs 60kB, as it fits into the F103C8 with only 20kB.

About using CubeMX to build my own F303RC version, I was thinking more in the direction of PlatformIO in VSCode.
I know both but prefer PIO (just a personal opinion).
Of course configuring the MCU and it's peripherals is easier/more intuitive in CubeMX.
Once I've the basic code generated by CubeMX, I use this source files to further develop in PIO.

I already started the F303RC PIO project, but still have too many errors to sort out.

parezj, 2022-01-27:
Hi Janjos0,

There will be problem with RAM for sure. Every build for each MCU has unique configuration. For example, every configuration tries to use as much RAM as possible, so in F103 has 10 kB and F303 has 44 kB, statically. You can see it in tables in paramaters section.

Thank you for nice tip PlatformIO in VSCode. I love VSCode and I will definitely check this soon.

janjos0, 2022-01-28:
Hi Jakub,

I understood the RAM size problem. After reading again the documentation, more specifically the limits table, I see now that you squeezed every byte out of each MCU type available RAM.
I didn't pay attention to this in the past and thought the buffer sizes used in the firmware were the same throughout the different MCU builds and the only change were for instance the peripherals speed.

I haven't yet been successful in my trial to build the F103C8 version in PlatformIO. The last compile errors I'm getting are:
"... section `._user_heap_stack' will not fit in region `RAM'"
"... region `RAM' overflowed by 560 bytes"
I try the F130C8 version first, because I have several BluePills to then test the bin file once (if) I succeed. :-)

I suppose the GUI app. also takes into account the type of MCU it is connected to and its limits, and just compiling the firmware for another MCU won't help without also modifying the app.

Anyway, because I like your project so much I will keep on trying the import into PIO, every time I have some spare time.

parezj, 2022-01-28:
Hi janjos0,

You don´t need to care about heap at all. Heap size is exactly 0 bytes. Everything is allocated statically. Try to set heap size to 0 or try to lower macro EM_DAQ_MAX_MEM value in file cfg_f103c8.h. This is the main buffer size which is filling all the available space in RAM. 

Yes, Qt GUI app is universal. After successfull connection to target, first commands read limits and set-up everyting.

Thank you very much again, I really appreciate it. Please try it, I hope you succeed soon :) I am here for your questions