parezj/EMBO #23: How to make such a device?

Ghost-Karpovtsov, 2024-08-23:
Good day everyone. I immediately apologize for the machine translation :( Has anyone actually replicated this device? How were the input circuits organized (as I understand it, more than 3.3 volts cannot be supplied to the inputs. Some additional circuits are needed for real use.

msalat, 2025-09-01:
Example frontend from scoppy with voltage divider and opamp: https://oshwlab.com/fruitloop57/scoppy-oscilloscope-analog-front-end-5

The frontend is for Raspberry Pi Pico, and with Wifi you're isolated, but...

if you're using one of the embo boards, measuring a circuit with a switching power supply as power source: be careful, "floating" due to "Safety and EMI/RFI suppression film capacitors". → ADuM1201

msalat, 2025-09-03:
... or a optocoupler for each rx+tx with digital out. Or a board with ch340 & optocouplers.

PS: the frontend 5 mentioned above has a op amp with 0.6 V / µs, → 5.5 µs for 3.3 V . For square wave signals I would take ×10, → 55 µs = 18 kHz. 
The Raspberry Pi Pico (→scoppy) has 500 kSa/s, the STM32F303 has ~5 MSa/s. → perhaps better OPA2340 op amp = 6 V / µs.